ros-industrial/staubli_experimental #7: gazebo: give 'remap' launch arg a more descriptive name

gavanderhoorn, 2016-04-01:
The `*_gazebo.launch` files currently expose a `remap` argument that, when `true`, remap the default topics from the `gazebo_ros` plugin (and friends) to the ROS-Industrial driver spec compliant ones.

`remap` is too generic a name, and should be changed into something which more clearly conveys its purpose / effect.

gavanderhoorn, 2016-04-01:
Launch files in `rx160_gazebo` and the to be introduced `tx90_gazebo` packages (see #6) are affected.

nilsmelchert, 2018-11-19:
Good idea. Besides renaming "remap" to something more meaningful, I think using the ROS-Industrial driver spec compliant topics by default would be a good idea. Since this is a ROS-Industrial repo, it makes a lot of sense to work with the industrial convention in gazebo by default.

nilsmelchert, 2018-11-19:
What about extending the argument "remap" to something more meaningful like "ros_remapping", "ros_remap", "ros_i_remapping" or "ros_i_remap" and set the default value to true.

nilsmelchert, 2018-12-07:
So if I did a PR for this, would you consider merging it? Or is there a reason not to do these changes?