ros-industrial/staubli_experimental #13: val3_driver: readme needs to be updated after merge into main repository

gavanderhoorn, 2016-07-11:
The [current version](https://github.com/ros-industrial/staubli_experimental/blob/f2259556476990aae2d5340daa60484bad38d871/staubli_val3_driver/README.md) of `README.md` needs to be updated:

 - refers to non-existent directory: `val3_driver`
 - instructs user to clone from `ocadotechnology` fork instead of main repository
 - refers to non-existent `val3_server-devel` branch

gavanderhoorn, 2016-11-13:
@MuriloMartins: do you think you can take care of this one? Use of this repository is increasing, and I'd rather people clone the main repository than the `ocadotechnology` fork (fi).

MuriloMartins, 2016-11-14:
@gavanderhoorn please see PR #17.

gavanderhoorn, 2016-11-15:
Fixed in #17.

gavanderhoorn, 2016-11-15:
Thanks @MuriloMartins for the fix.

Bit early, but: :beers: :)

closed 2016-11-15 by gavanderhoorn