ros-industrial/staubli_experimental #15: val3_driver: create ROS wiki page

gavanderhoorn, 2016-07-11:
The `README` already contains quite some information, but is not the first place most users would look to for instructions on installation and usage.

It would be good to expand the ROS wiki page for `staubli_val3_driver` to include things such as minimum version of VAL3 supported, compatible controllers, information on compatible manipulators, etc.

See [fanuc_driver](http://wiki.ros.org/fanuc_driver?distro=indigo) for an example of such a page.

gavanderhoorn, 2016-07-11:
Alternatively, we might want to invest in writing Sphinx-based documentation for the `staubli` packages.

That would have a nr of advantages over the ROS wiki (versioning with source, easier on resources, target multiple output formats with a single source, responsive layout, etc). A disadvantage is less integration (and visibility?) with the ROS wiki.

See also ros-industrial/fanuc#148.

gavanderhoorn, 2018-06-19:
@nilsmelchert: is this perhaps something you would be interested in?

The current page is [wiki/staubli_val3_driver](http://wiki.ros.org/staubli_val3_driver).

You could take a look at the [fanuc_driver](http://wiki.ros.org/fanuc_driver?distro=indigo) one mentioned in the OP (see also the [installation tutorial](http://wiki.ros.org/fanuc/Tutorials/hydro/Installation)) and the Motoman [installation tutorial](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer).

If you've run into any issues while setting up `staubli_val3_driver`, it might be nice to document those on a `Troubleshooting` page similar to the ones for `fanuc_driver` ([here](http://wiki.ros.org/fanuc_driver/Troubleshooting)) and `motoman_driver` ([here](http://wiki.ros.org/motoman_driver/Troubleshooting)).

nilsmelchert, 2018-11-19:
Sure. I totally forgot about this Issue. I will take care of it.