ros-industrial/staubli_experimental #18: Migration of the staubli tx60

nilsmelchert, 2018-03-06:
I will add the tx60 robot to this rebository locally on my machine. If you wish, we could migrate it to the existing "staubli_experimental"-Repository?

gavanderhoorn, 2018-03-06:
Hi.

Contributions of new robot support packages are always welcome, yes. A PR contributing support for the TX60 series would be much appreciated.

Are you basing your work on one of the existing packages?

In that case, please consider using some of the newer 'best practices' for xacro macros shown in [this file](https://github.com/ros-industrial/fanuc_experimental/blob/6462dd3154aefe96252a4997ae2f426c23f84e4a/fanuc_m20ib_support/urdf/m20ib25_macro.xacro)? Specifically the `base`, `flange` and `tool0` frames, and the use of `${radians(..)}` instead of manually converting degrees to radians in joint `limit` attributes.

Also: as Staüblis are somewhat flexible with regards to maximum payload and joint torque limits, a note similar to the one we have for the RX160 ([here](https://github.com/ros-industrial/staubli/blob/c6b80b00610e21833d0d4d45c5ce0ee803a2167d/staubli_rx160_support/package.xml)) would be nice to include.

nilsmelchert, 2018-03-06:
Yes, I was going to base my work on the TX90. 

- By newer 'best practices' you mean, that the joint definitions are right underneath the definition of a certain robot link?

- What is the purpose of the link 'base' when 'base_link' is already definied?

- What is the purpose of flange? Isn't 'tool0' already the attachment point for tools, which are connected to the endeffector?

I will try to follow the given conventions as good as i can.

gavanderhoorn, 2018-03-06:
re: `base` and `flange`: please see [Coordinate Frames for Industrial Manipulators](https://github.com/gavanderhoorn/rep/blob/coord_frames_ind_robots/rep-xxxx.rst) for now. It's a draft, but there will be a REP about these frames, documenting their use and why we feel those make sense.

gavanderhoorn, 2018-03-06:
@nilsmelchert wrote:
> Since I am already working with the TX90 Robot, I could also update these 'newer conventions' to that package if needed.

that is a good idea, but let's focus on the new additions first. If/when we get the updates sorted out with the TX60, we can take a look at the TX90 variants.

nilsmelchert, 2018-03-09:
I forked the staubli_experimental repository and created a new branch. I am almost done, migrating the tx60 series. But there is a little thing, that I am still uncertain about:

> In that case, please consider using some of the newer 'best practices' for xacro macros shown in this file? Specifically the base, flange and tool0 frames, and the use of ${radians(..)} instead of manually converting degrees to radians in joint limit attributes.

In the package `fanuc_m20ib_support` both, `tool0` and `flange` depend on `link_6`.
Wouldn't it make more sense, if the frame `tool0` depends on `flange`?

gavanderhoorn, 2018-03-09:
@nilsmelchert wrote:
> In the package `fanuc_m20ib_support` both, `tool0` and `flange` depend on `link_6`. Wouldn't it make more sense, if the frame `tool0` depends on `flange`?

Yes, it would.

The REP already specifies it that way, but this was one of the discussion points. There are *some* (ref needed) robot brands that have `tool0` defined relative to the intersection of `joint_4`, `joint_5` and `joint_6` (or the `link`s of those, in a urdf). With those in mind, I allowed `flange` and `tool0` to be siblings and both children of `link_6`.

Following the discussion that has now changed, and now `flange` is the parent of `tool0`. Even for robots where this might not be 'physically' the case (we're talking mathematical/virtual constructs here, so in quotes), we still model it like that.

So to summarise: yes, make `tool0` a child of `flange`. The `fanuc_m20ib_support` package should be updated.

nilsmelchert, 2018-03-09:
Ok thanks. 

I was also wondering about the orientation of `flange`. It is said, that

> In contrast to tool0, this frame shall always be oriented such that it complies with REP 103 [1]. Positive X must always point away from the last link (ie: in the 'forward' direction for a world-aligned robot model).

I am not quiet sure about in what direction i should position the `flange` frame. Even though it says `Positive X must always point away from the last link`, it doesn't seem to be clearly defined to me.

gavanderhoorn, 2018-03-09:
That particular phrasing is currently under review as well, and I will probably add a diagram to clarify it.

For a robot where the last link already follows `REP-103` (so zero-pose has robot wrist aligned with `x+`), `flange` is expected to be coincident with `link_6` (ie: identity transform). For a robot that in its zero-position is straight-up (ie: points to the sky, such as a KUKA IIWA or a Fanuc M-430iA), and all `link`s only have z-axis translations between them, `flange` should be -90 degrees rotated over y, so the `+x` axis now points straight-up as well. See [m430ia2f_macro.xacro](https://github.com/ros-industrial/fanuc/blob/21d16189c3640c7f161a56504edec4db26eb7735/fanuc_m430ia_support/urdf/m430ia2f_macro.xacro#L143-L149) for an example.

As ROS (or: TF) attaches child links to the `+x` axis by default (`REP-103` again), this should make attaching child links (and sub-assemblies) more straightforward, as they would now be correctly mounted against the robot flange.

If we were to use `tool0` as an EEF attachment point, which has `+x` typically pointing 'down' (due to the `+90` rotation over y), we'd have to undo that rotation or the EEF subassembly will be sticking out 'to the side' of the robot.

The idea is that when switching robot variants (in a top-level / composite xacro fi) attachment of the EEF subassembly will always be the same: along the `x+` axis.

nilsmelchert, 2018-03-09:
Ok thanks.
So I changed the orientation of `flange`. 
Since `tool0` is a child of `flange`, its orientation has to be changed again, so that its z-axis points upwards when the robot is in its all-zero-position right? In other words `tool0` should be -90 degrees rotated over y?

gavanderhoorn, 2018-03-09:
`tool0` should follow Staübli's definition. If I recall correctly, Staübli has toolframes with `z+` sticking out of the robot flange, so `tool0` would need a `+90` degree rotation over y (provided that `x+` of `flange` is now world-aligned).

The zero-pose of a `tx60` is similar to that of the tx90, is it not?

nilsmelchert, 2018-03-09:
> The zero-pose of a tx60 is similar to that of the tx90, is it not?

That's correct.

I think, I am done now. The only thing I am still struggling with, is the update of the information of the package.xml-file:
- can't find the effort limits of the robots
- don't know the current version of instruction manual

Are you able to add the missing information?

gavanderhoorn, 2018-03-09:
What was your source for the joint limits and kinematic structure of your support package? Just list that.

As to the effort limits: they should be listed in the documentation that came with your robot / can be found on Staübli's website.

gavanderhoorn, 2018-03-09:
Btw: the inertias and masses in the xacros of the tx90 are estimates from SolidWorks. Did you update those for the tx60?

If you don't have data for your robots, then please leaves those elements out, as reusing them is probably not a good idea.

nilsmelchert, 2018-03-09:
> What was your source for the joint limits and kinematic structure of your support package?

I used [this](https://www.staubli.com/en-de/file/5285.show) Information for the joint limits, max. velocities and for the kinematic structures. It is the data sheet from the official website from Stäubli.

> As to the effort limits: they should be listed in the documentation that came with your robot / can be found on Staübli's website.

Concerning the effort limits, I will see if i can find them and then add the information the the package.xml-file.

> Btw: the inertias and masses in the xacros of the tx90 are estimates from SolidWorks. Did you update those for the tx60?

I have not updated the inertias and masses of the tx60 series yet. But I will figure them out using solid wokrs.

---------------------------------

I will get these things done by monday and get back at you then.

gavanderhoorn, 2018-03-09:
The datasheet you link is for the `TX2` line. I haven't checked, but are those identical to the `TX` line? If not, we would need to reflect that in the package name.

nilsmelchert, 2018-03-09:
To be honest, since I am german, I used the [german version](https://www.staubli.com/de-de/file/5285.show) of the datasheet.
The data sheet of the `TX2` line seems to be identical to the `TX` line.

Where do you want me to mention that?

nilsmelchert, 2018-03-09:
Did you use the [SolidWorks to URDF Exporter](http://wiki.ros.org/sw_urdf_exporter) to calculate the inertia, mass, and center of mass of certain robot links?

gavanderhoorn, 2018-03-09:
If the TX2 is mechanically, kinematically and dynamically identical to the TX, then we don't need to create a separate package, but can just refer users looking for TX2-60 support to the `staubli_tx60_support` package. This can be noted in `package.xml` and the readme.

If there are differences, and you've actually been modelling the TX2-60, then it might make sense to change the name of the package to `staubli_tx60_2_support`.

gavanderhoorn, 2018-03-09:
@nilsmelchert wrote:
> Did you use the [SolidWorks to URDF Exporter](http://wiki.ros.org/sw_urdf_exporter) to calculate the inertia, mass, and center of mass of certain robot links?

No, or at least, not me personally. The `TX90` and `RX160` packages were created partly by contributors (@MuriloMartins fi) and they used SolidWorks without any plugins afaik.

gavanderhoorn, 2018-03-09:
If you don't use these packages for dynamic simulations, you may not need to add those elements. It would be nice to have them, but I can imagine that if you don't use it, it's going to be hard to verify that you've got something sane.

In that case, it would also be ok to submit your PR without any dynamics properties in the xacros.

Those can always be added later.

nilsmelchert, 2018-03-09:
> If there are differences, and you've actually been modelling the TX2-60, then it might make sense to change the name of the package to staubli_tx60_2_support.

No, I am sure, that i modeled the tx60 series. There is a mistake in the english version of the datasheet naming the robot version. I just called stäubli and they confirmed it.

> If you don't use these packages for dynamic simulations, you may not need to add those elements. It would be nice to have them, but I can imagine that if you don't use it, it's going to be hard to verify that you've got something sane.

I will use the gazebo simulation and will add the dynamics information for the tx60 series.

nilsmelchert, 2018-03-12:
What material is good to chose to get the inertia matrix and the mass of the robot link?
Pure steel seems to be a little bit too heavy.

gavanderhoorn, 2018-03-12:
Yes, that is one of the difficult choices when using SolidWorks for this, unless the mfg has assigned materials.

I'm not sure what the best approach would be.

You could perhaps see whether you can find some publications about system identification of a Staubli TX60. I know there are for TX90.

nilsmelchert, 2018-03-12:
There is no publication of a system identification of the tx60 series online, that i found. 
My suggestion would be to use the identified Mass of the tx90 and calculate the density of certain links. Since I the tx90 and tx60 series are very similar, that should lead to satisfying results (at least at first).
What do you think?

gavanderhoorn, 2018-03-12:
Sounds like a good first approach, yes.

You could even try and see whether Staübli would be willing to provide you with the link masses. Just don't mention you're going to do any sort of dynamic simulations with that information. They don't like that too much.

nilsmelchert, 2018-03-12:
What approach did you use for the rx160? 

I will see if i can get any information from the saubli support.

gavanderhoorn, 2018-03-12:
> What approach did you use for the rx160?

@MuriloMartins and his colleagues used SolidWorks with some guesses for the materials. I only modelled the kinematics.

See ros-industrial/staubli#15.

nilsmelchert, 2018-03-12:
@MuriloMartins entered the same inertias and masses for the whole tx90 series, which can't be right. 
I just talked to staubli and they won't give me the inertias and masses of the robot links unless I sign a contract, that I won't publish them.

As a first approach, I will calculate the inertias as I mentioned before.

What will be the next step when I am done? Can we add my "staubli_tx60_support" package to the repository?

gavanderhoorn, 2018-03-12:
> I just talked to staubli and they won't give me the inertias and masses of the robot links unless I sign a contract, that I won't publish them.

well, I expected that. And I probably wasn't clear: talking about inertias is immediately a 'red flag' to Staübli. But asking for the masses of the links could perhaps have been general enough to not make Staübli immediately reach for an NDA.

> What will be the next step when I am done? Can we add my "staubli_tx60_support" package to the repository?

If you submit a PR against the repository with your changes, we'll start a proper review. If that is approved, it can be merged.

nilsmelchert, 2018-03-13:
> well, I expected that. And I probably wasn't clear: talking about inertias is immediately a 'red flag' to Staübli. But asking for the masses of the links could perhaps have been general enough to not make Staübli immediately reach for an NDA.

I was actually only asking for the masses. But I will anyways to a system identification for the tx60 in the future and can add this information later on to the repository.

nilsmelchert, 2018-03-13:
Seems like @MuriloMartins assumed the density of all robot links of the tx90 was exactly 1000 kg/m^3. 

I will make the same assumption for the tx60 series.

nilsmelchert, 2018-03-13:
> If you submit a PR against the repository with your changes, we'll start a proper review. If that is approved, it can be merged.

I think the `staubli_tx60_support` package is ready to be reviewed.

gavanderhoorn, 2018-03-13:
Then I suggest you submit a pull request.

nilsmelchert, 2018-03-14:
> Then I suggest you submit a pull request.

Done.

Let me know if I can help.

nilsmelchert, 2018-03-15:
Are you the one who is going to review my work?

gavanderhoorn, 2018-03-15:
Yes.

I will do it, just have a lot of other things to do as well.

I've not forgotten your PR :)

gavanderhoorn, 2018-03-21:
Now that #19 has been merged, we can close this one.

closed 2018-03-21 by gavanderhoorn