ros-industrial/staubli_experimental #20: Removal of deprecated syntax for the staubli_tx90_support package

nilsmelchert, 2018-03-21:
I was making a couple of changes on my private repository, deleting some deprecated syntax of the `staubli_tx90_support` package like:

- Replacing xacro.py with xacro and adding --inorder

- Unify the naming conventions of `xmlns:xacro="http://wiki.ros.org/xacro"` in the xacro files

- Removing the `--shutdown-timeout 1` entry from `joint_state_controller.yaml`

Should I make another pull request in order to merge those changes to `ros-industrial/staubli_experimental`?

gavanderhoorn, 2018-03-21:
Yes, that would be great to have as a PR.

We could also look at adding the `flange` and `base` frames to the `rx160` package.

gavanderhoorn, 2018-03-21:
As to branching, we can do that, but it might be nice if we could see whether we can come up with a way to fix things for Kinetic while maintaining compatibility with Indigo.

> Removing the `--shutdown-timeout 1` entry from `joint_state_controller.yaml`

I think you mean in some launch file, as `yaml` files don't typically contain these things.

nilsmelchert, 2018-03-22:
> Yes, that would be great to have as a PR.
> 
> We could also look at adding the flange and base frames to the rx160 package.

Done. I did a PR for `staubli_experimental` #22 and one for the `staubli` package (see [#25](https://github.com/ros-industrial/staubli/pull/25) at the `staubli` package).

nilsmelchert, 2018-03-22:
I think you mean in some launch file, as yaml files don't typically contain these things.

Correct. I meant the `control.launch` file from the gazebo packages.

closed 2018-04-09 by nilsmelchert