ros-industrial/staubli_experimental #21: When will staubli_experimental be migrated to the staubli repository?

nilsmelchert, 2018-03-21:
When will [staubli_experimental](https://github.com/ros-industrial/staubli_experimental) be migrated to the [staubli](https://github.com/ros-industrial/staubli) repository? What kind of testing has to be done in order to to that?

gavanderhoorn, 2018-03-21:
The package itself will not be migrated, as that doesn't make much sense, but individual packages in the repository may be migrated after they've seen 'enough use'. That is a somewhat qualitative (and admittedly vague) criterium, but unfortunately there's not a very concrete way to define when these packages are 'good enough'.

Validation and verification wrt real robots can be done, but takes time, and even then doesn't guarantee that the pkg is completely fault free and/or 'done'.

As to the pkgs that have been in this repository for the past year (TX90 pkgs mostly): those can be migrated at our earliest convenience (but probably after we address the points you raised in #20).

gavanderhoorn, 2018-03-22:
As this is not really an issue with the pkgs in this repository, I'll close this issue.

You're more than welcome to keep commenting on it though.

closed 2018-03-22 by gavanderhoorn