ros-industrial/staubli_experimental #25: val_3 driver: rough and clipped trajectories

nilsmelchert, 2018-04-24:
After working a while with `ros kinetic` and the `val_3 driver` in the `staubli_experimental` package, I figured that the executed trajectories of the real robot are not running very smoothly. Especially in the end of an executed path the movement is quite clipped.
Did also experience the same thing? Could there be a problem with the driver? Or is there a problem with the `ompl` library?

gavanderhoorn, 2018-04-24:
The driver could always play a role (it hasn't been tested *very* extensively, partly why it is in an experimental repository), but if you're using Kinetic and MoveIt, you may be running into ros-planning/moveit#416 or related issues.

See ros-industrial/kuka_experimental#126 for a possibly related issue, and some links to relevant other issues.

I'm not saying this is the cause of what you're seeing, but it would be good to check this.

One way to see whether Kinetic OMPL is doing strange things is to generate a trajectory in some other way, time-parameterise it yourself and then send it to the driver directly.

> Especially in the end of an executed path the movement is quite clipped.

I'm not sure I completely understand what you're describing here. What do you mean with 'clipped'?

nilsmelchert, 2018-04-25:
> I'm not sure I completely understand what you're describing here. What do you mean with 'clipped'?
By clipped I meant very jerky. 

Yes [ros-planning/moveit#416](https://github.com/ros-planning/moveit/issues/416) seems to describe the behaviour. The second comment in [this answer](https://github.com/ros-planning/moveit/issues/416#issuecomment-273718781) could be a reason for the jerky trajectory execution.

gavanderhoorn, 2018-04-28:
According to the latest comments on ros-planning/moveit#416, the issue with OMPL is now fixed and is going to be backported to Kinetic.

nilsmelchert, 2018-04-28:
Thank you very much! I will close this Issue now.

nilsmelchert, 2018-04-28:
When will I be able to update the fix for this Problem? Will it be updatable in the apt-tree or would I have to apply the changes myself with some kind of patch which will be published?

What are the next steps?

gavanderhoorn, 2018-04-28:
The fix first needs to be verified. Then it needs to be backported to Kinetic. MoveIt/OMPL will need a new release, and that will need to be synced to the public repositories.

It could take some time. Building OMPL from source on Kinetic -- and thus MoveIt as well -- will make it available earlier to you.

Note also btw that it's not certain that this is the cause of what you describe. It's a good candidate, but there are no guarantees.

gavanderhoorn, 2018-04-28:
I recommend you keep a close eye on ros-planning/moveit#416.

gavanderhoorn, 2018-06-19:
Have you seen any improvement @nilsmelchert with the release of the new MoveIt+OMPL?

gavanderhoorn, 2018-06-22:
MoveIt & OMPL on Kinetic should have all the updates. No need for a source build.

nilsmelchert, 2018-06-22:
Ok. I will test if there is any improvement and then post the results.

gavanderhoorn, 2018-06-22:
One thing to check btw is that the `reach` and `leave` setting for each trajectory point in the driver is set to really low values, so the controller cannot do much blending.

That could also be the cause of some jittering.

See [this diff](https://github.com/ros-industrial/staubli_experimental/compare/indigo-devel...MPJansen:684da890) for another user that experimented a bit with that.

Ideally blending should not be used, but if all else fails, it might be something to look into.

gavanderhoorn, 2018-11-02:
@nilsmelchert: has this improved at all for you?

closed 2019-05-09 by nilsmelchert