ros-industrial/staubli_experimental #26: What is the function of this package now?

DongMuJi, 2018-06-19:
Hi 
Now I want to use this package. But I do not konw what is the function of this package now.
I also see a stack named staubli. What is different between staubi_experimental and staubli?
Thank you very much!

gavanderhoorn, 2018-06-19:
@DongMuJi: @nilsmelchert is exactly right.

gavanderhoorn, 2018-06-19:
We should start migrating the packages in `staubli_experimental` to `staubli` though.

gavanderhoorn, 2018-06-19:
@nilsmelchert wrote:
> Besides that, there has also been a hardware interface ([staubli_val3_driver](https://github.com/ros-industrial/staubli_experimental/tree/kinetic-devel/staubli_val3_driver)) delevoped by @muhrix in order to communicate with the "real" robots.

One nitpick: let's not use the name "hardware interface" for `staubli_val3_driver`, but let's call it a "driver".

`hardware_interface` is a package in `ros_control` and that works quite differently from how `staubli_val3_driver` works.

nilsmelchert, 2018-06-19:
> We should start migrating the packages in staubli_experimental to staubli though.

What would be the next steps to make this happen?
Since I am working with the tx60 and the tx90 robots (in gazebo simulation mode and with the staubli val3 driver for the CS8C Controller), I can confirm, that I have not had any problems with any of the packages.
The val3 driver works perfectly fine.

gavanderhoorn, 2018-06-19:
For all packages in `staubli_experimental` we can migrate them.

There is no fixed 'period' of time as it also depends on usage (ie: a bunch of pkgs sitting in a repository unused for 6 months is not a good indicator of stability, regardless of whether they see PRs or issues being reported).

But all packages in this repository have seen sufficient usage.

> What would be the next steps to make this happen?

We'd need to start opening PRs migrating the packages. I have a process documented, but it's not publicly accessible right now. I could make it available, but there is a bit of a debate (see ros-industrial/ros_industrial_issues#49 fi) whether to do that with all history or just do a 'current-state' import.

nilsmelchert, 2018-06-19:
Well then I would just start with the Tx90 support and gazebo packages.

If I get you right, there is a debate about moving the hardware drivers of ros industrial robots into an isolated repo. 
I actually like that the location of the hardware driver. Since the Issue you mentioned is still open, I would recommend to also move the staubli_val3_driver to the staubli repo for now.

Are you ok with that?

gavanderhoorn, 2018-06-19:
@nilsmelchert wrote:
> If I get you right, there is a debate about moving the hardware drivers of ros industrial robots into an isolated repo.

the debate is about whether to move things *with* git history (ie: commits) or without. ros-industrial/ros_industrial_issues#49 touches on that, which is why I mentioned it.

Let's continue discussion of migration of pkgs in #27.

gavanderhoorn, 2018-06-19:
@DongMuJi: as your question is not a software-related issue, but more a usage one, I'm going to close this here for now.

If you still have questions you can keep commenting on this issue though.

closed 2018-06-19 by gavanderhoorn

gavanderhoorn, 2018-06-21:
`staubli_val3_driver` supports joint position control only. No velocity or torque control.

It also only reports joint positions, but that could be extended to include velocity and effort if the VAL3 code is extended.

As to the warnings and errors: can you please show the exact commands that you are using to start things? What do you mean exactly with "robot module"?

Please also provide some more details on your OS, ROS version and how you installed everything. We cannot guess these things.

And please try to avoid posting images that show console output: it's all text, so please just copy-paste it instead of making photos of monitors.

jacktnho, 2018-10-22:
> `staubli_val3_driver` supports joint position control only. No velocity or torque control.
>
> It also only reports joint positions, but that could be extended to include velocity and effort if the [...]

Hello!
May i ask if the current package staubli_val3_driver support controlling the digital I/O ports of the CS8 controller? 
many thanks!

nilsmelchert, 2018-10-22:
As far as I know, the IO is not implemented so far. Nor are the effort values of the joint torques. 
You are of course very welcome to create a PR for those new features.

simonschmeisser, 2019-07-23:
Highjacking this old question and knowing your objections @gavanderhoorn ;) ... how would I toggle the valve on `ValveIO\valve1` from VAL3?

gavanderhoorn, 2019-08-01:
If you are asking how you'd do that using `staubli_val3_driver`, then I believe the answer would be: you'd have to add it, as there is no support for that right now.

I believe you've already done that in your fork though.

My recommendation for any production system would still be to use a proper fieldbus. Perhaps even `modbus` (for which several ROS packages exist).