ros-industrial/staubli_experimental #27: Migration of pkgs to non-experimental repository

gavanderhoorn, 2018-06-19:
Triggered by discussion in #26.

nilsmelchert, 2018-06-19:
So in order to create PRs for the staubli repository, I just fork the current master branch from the `staubli` repository and add the packages I want to migrate from the `staubli_experimental` repository manually?

gavanderhoorn, 2018-06-19:
@nilsmelchert wrote:
> I actually like that the location of the hardware driver. Since the Issue you mentioned is still open, I would recommend to also move the staubli_val3_driver to the staubli repo for now.

Tbh it doesn't make much sense to me to migrate the driver to `staubli` and *then* migrate it again to its own repository.

The issue is still open, as the migration/extraction hasn't happened yet, but the decision has already been made. See [gavanderhoorn/staubli_val3_driver](https://github.com/gavanderhoorn/staubli_val3_driver) for a prototype for `staubli_val3_driver`.

gavanderhoorn, 2018-06-19:
@nilsmelchert wrote:
> So in order to create PRs for the staubli repository, I just fork the current master branch from the `staubli` repository and add the packages I want to migrate from the `staubli_experimental` repository manually?

at a high level, that is what needs to happen yes. But in order to make this a bit more of a structured process, and to make sure we don't completely disconnect things, there is a procedure to follow. As I wrote in https://github.com/ros-industrial/staubli_experimental/issues/26#issuecomment-398355070, I have that procedure documented, but it's not publicly accessible right now.

The reason for that is the discussion I mentioned earlier (ie: with or without history).

Summarising: it's slightly more complex than just creating PRs and `git rm ..`.

nilsmelchert, 2018-06-19:
Alright. So are you going to make the process of migrating certain packages public? 
How can I help?

gavanderhoorn, 2018-06-19:
I'm really rather busy right now, so I'll see if I can find 20 mins somewhere to make it available. It won't be today though I'm afraid.

gavanderhoorn, 2018-06-19:
@nilsmelchert wrote:
> How can I help?

and :+1: :100: for offering to help here btw, really appreciated.

nilsmelchert, 2018-06-19:
No worries. Let me know, when you want me to start.