ros-industrial/staubli_experimental #32: Driver skips points in between trajectory

gonzalocasas, 2019-05-05:
The motion control task of the driver [skips trajectory points](https://github.com/ros-industrial/staubli_experimental/blob/kinetic-devel/staubli_val3_driver/val3/ros_server/motionControl.pgx#L51) when they match the current configuration.

Isn't there a missing check to ensure this only happens for the first point of a trajectory?
At least, that seems to be the [condition implemented on the ABB driver](https://github.com/ros-industrial/abb/blob/kinetic-devel/abb_driver/rapid/ROS_motion.mod#L59) (which is conceptually the same but also checks `current_index  == 1`)

gavanderhoorn, 2019-05-05:
That looks like an incorrect implementation indeed: there's no way to guarantee that `herej()` will be called at the right time -- most likely it will be called at the wrong time, as you note.

nilsmelchert, 2019-05-09:
Could this problem cause the jerky trajectories that were mentioned in #25 ?

gonzalocasas, 2019-05-09:
It does cause trajectories to execute weirdly (depending on several factors, of course). We tested generating trajectories in MoveIt and then creating val3 code out of them and that works perfectly, so I would rule out MoveIt/OMPL as the source of jerkiness.

gavanderhoorn, 2019-05-09:
@gonzalocasas: did you already check whether removing that line improves things?

Should be relatively easy to check, just comment the line.

gonzalocasas, 2019-05-10:
@gavanderhoorn yes, I confirm. Removing the line indeed fixes the issue.

gonzalocasas, 2019-05-10:
I can send a PR for it as is (just removing the line), although, I wanted to check first whether it was still meaningful to do this check but move it to the execution loop, so that it actually does what it was intended to do. But I can also send the removal-line PR for now and then send the other if I see it makes sense... What would be preferable from your perspective?

gavanderhoorn, 2019-05-10:
Let's just remove the line.

I'm not sure it makes sense to add it back somewhere else, as `here()` will always have a delay / be desynced with the current state of the robot.

It *could* be that pruning points might make sense / improve performance when very dense trajectories are streamed in. In that case `here()` + a margin and filtering for that could be something to consider.

I don't have any way to test that, but it could be something to keep in mind for the future.

commit 1ff47df2 (2019-05-20): Merge pull request #33 from gramaziokohler/fix/point-skipping

fix #32: do not skip points based on herej()
commit 63cd22bb (2019-05-20): fix #32: do not skip points based on herej()

A longer description of the problem is available on the github issue,
but the basic problem is that this is not evaluated at the correct time
(i.e. when the move is about to be executed), but at the time the move
is being queued, this means having the robot on a paused state already
triggers this problem.
closed 2019-05-20 by gavanderhoorn