ros-industrial/staubli_experimental #34: tx2 compatibility

simonschmeisser, 2019-07-11:
We are currently discussing about having some initial experiments/trial with stäubli in about a week and the customer seems to have a tx2-60. So I'm wondering if you expect this driver to be compatible to cs9 as well? Is there anything special to watch out for?

I'll add a pr for the support package later on of course

gavanderhoorn, 2019-07-11:
@gonzalocasas might respond later, but for now see [299054](https://answers.ros.org/question/299054) on ROS Answers.

Summarising: the current code is not compatible with CS9 controllers. Afaik, @gonzalocasas does have a version that works, but it hasn't been upstreamed yet.

simonschmeisser, 2019-07-11:
Thanks @gavanderhoorn  :) @gonzalocasas I could also provide some help or testing if needed. A code-dump of the current state would also be totally fine, no need for perfection ... yet ;)

gonzalocasas, 2019-07-11:
@simonschmeisser indeed, as @gavanderhoorn points out, we have a fork of the driver for the cs9 controller (we use 2 tx2-60l arms at the moment).

The only reason why I haven't pushed that yet is that ideally the code should work with both cs8 and cs9 but maybe for now, it's fine to branch and ignore cs8 support on that branch.

gonzalocasas, 2019-07-11:
I will push a branch in the later today so that you can give it a try.

For the most part, the driver seems to behave well, we are still testing it, and just today we noticed problems with very long trajectories, but nothing unsurmountable.

simonschmeisser, 2019-07-11:
Cool, much appreciated! :100: 

What does "very long" mean? Number of trajectory points? Duration of movement? Distance between points?

gavanderhoorn, 2019-07-11:
There is a known issue with the default IRC (which is used in the `staubli_val3_driver` launch files) where trajectories with a duration over approx 4.6 hours cannot be encoded correctly due to the use of `float` instead of `double` for the time stamp field.

See https://github.com/ros-industrial/motoman/issues/244 for a description in the `motoman` repository.

It's an actual problem, but to be honest, trajectories of over 4 hours are not the kind of trajectories I would expect in a "standard" ROS application.

I'm not sure whether that is what @gonzalocasas encountered, but it *is* a limitation of the use of `float`s.

gonzalocasas, 2019-07-11:
Not that kind of long ;) We encountered issues when the number of points in a trajectory is large. We didn't test exactly where it breaks, but today we tried to send a 798-points trajectory and it stopped somewhere at around 127 points or so... I ran the same trajectory again, and the problem repeated but then it stopped at around 107 approx. I did not have time to debug any further, so I have no clue where the problem comes from, but since everything else has been working fine, and the IRC (industrial robot client) reports sending all 798 points correctly, I assume we're hitting some kind of limit on the driver side.

gavanderhoorn, 2019-07-11:
>  I assume we're hitting some kind of limit on the driver side.

"on the server side" you mean.

The VAL3 code includes all sorts of processing of points and adding them to internal queues, so perhaps this is a memory issue. There certainly isn't an upper bound on trajectory length in a streaming driver.

nilsmelchert, 2019-07-11:
@gonzalocasas If I understood you right, the syntax of the val3 language changed a bit?
So instead of creating a new branch, shouldn't we just create two subfolders for the `staubli_val3_driver`?
The hirachy would then look something like this:

For CS8C: `staubli_experimental/staubli_val3_driver/val3/cs8c`
For CS9C: `staubli_experimental/staubli_val3_driver/val3/cs9c`

Or is there a possible way of making the driver compatible?
I am also interested in that kind of topic, since we are also planning on working with a TX2 90 robot soon.

@gonzalocasas have you also already implemented the support packages for the tx2 60 series?

gonzalocasas, 2019-07-11:
> "on the server side" you mean.

yes, on the VAL3 server side.

> The VAL3 code includes all sorts of processing of points and adding them to internal queues

exactly my suspicion. Despite the name in the queue processing, (the queue functions are named push and pop), the points are not really popped as they are executed (the trajectory buffers are only cleared at the end), so it holds all the points in memory until trajectory is completed.

gavanderhoorn, 2019-07-11:
> exactly my suspicion. Despite the name in the queue processing, (the queue functions are named push and pop), the points are not really popped as they are executed (the trajectory buffers are only [...]

yeah, that would be a problem, and essentially turns it into a downloading driver implementation.

gonzalocasas, 2019-07-11:
> essentially turns it into a downloading driver implementation

Almost, yes. Not entirely thou, because the motion control task indeed starts moving as soon as there are points in the queue, and it moves forward through the queue without waiting for the full trajectory to be in memory, but since it does not release any memory while it can, it's not really able to cope with streams of many points. (Disclaimer: all of this is very speculative, as said, only faced this issue this afternoon and I have not had enough time to properly analyze it)

gonzalocasas, 2019-07-11:
@nilsmelchert val3 changed a bit, but only slightly. It's the UI-related functions that change, the move from text-based to html-based interface on the CS9 controller.

Having two folders is an option, but ideally, one code base that can cope with both controller versions would be better than having to copy the correct version to the controller. But I don't know if the language allows you to detect and do certain things conditionally, especially because one of the changes  is on a global type, which doesn't exist anymore ([type screen](https://github.com/ros-industrial/staubli_experimental/blob/kinetic-devel/staubli_val3_driver/val3/ros_server/ros_server.dtx#L269)). I have no clue whether that can be dynamically created or skipped

> have you also already implemented the support packages for the tx2 60 series?

yes, I will try to push that as well.

gavanderhoorn, 2019-07-11:
We could potentially ask Staubli to assist us here.

The driver is most likely not the only VAL3 application that has to be ported from CS8 to CS9 environments.

I have some contacts at Staubli, but if any of you have some as well I would encourage you to ask them.

nilsmelchert, 2019-07-11:
I have exhibited at the [hannover fair](I also exhibited at the fair this year.) this year and talked to staubli representitives. They told me, that all of val3 code should be compatible with the new cs9c controller (Which is obviously wrong).

Since I am about to purchase another robot, I am pretty sure staubli willing to help when it comes to migrating "old" val3 code. 

I will call them tomorrow and let you know the results.

gavanderhoorn, 2019-07-11:
Would be especially interesting to know whether it'd be possible to do this in a bw compatible way.

Duplicating the code and adapting it for the CS9 is of course possible, but as @gonzalocasas writes, ideally we avoid that, as it also duplicates maintenance effort.

nilsmelchert, 2019-07-11:
Yes that's of course what I meant. I will ask, if there is a guideline of making the val3 code compatible for both controller types.

gonzalocasas, 2019-07-11:
@simonschmeisser 👉 here's the branch with the CS9 changes: https://github.com/gramaziokohler/staubli_experimental/tree/kinetic-devel-cs9

@nilsmelchert in case it helps to illustrate the problem, the main question mark for me is how to do [this global variable definition](https://github.com/ros-industrial/staubli_experimental/compare/kinetic-devel...gramaziokohler:kinetic-devel-cs9#diff-d357ca6041db797257917a32bba9a96dL269)  in a way that works on CS9 controllers.

 As a side question: does any of you have a CS8 at hand to test?

gavanderhoorn, 2019-07-11:
No. Unfortunately I actually only have a CS9 myself (for a little while anyway).

nilsmelchert, 2019-07-11:
I have a cs8 controller available.

simonschmeisser, 2019-07-12:
@gonzalocasas thanks a lot, I'm glad I asked :smile:

gonzalocasas, 2019-07-12:
> I have a cs8 controller available... actually two 😉

great! as soon as we hear back from Staubli via @nilsmelchert, I will update the fork and if you could give it a go on the real CS8, I can test it on our CS9.

nilsmelchert, 2019-07-12:
So, I talked to two different technical support engineers and both told me that `"every single val3 application is compatible with both, the cs8 and cs9 controllers"`. 

> @nilsmelchert in case it helps to illustrate the problem, the main question mark for me is how to do this global variable definition in a way that works on CS9 controllers.

Concerning the global variables, I got the same answer, that they are definitly compatible. They said the only thing that changed significantly is the GUI of the hand controller.

Sorry I clouldnt 't help you out for now. Since I don't have a CS9 controller available myself, I will simulate one in the SRS2019 environment.

gavanderhoorn, 2019-07-12:
> They said the only thing that changed significantly is the GUI of the hand controller.

which is what @gonzalocasas also already wrote:

> the `staubli_experimental` repository is **not** compatible with CS9, but only due to the change from text interfaces to HTML-based that CS9 brings. Disabling the `screenUpdate` task brings it back into compatibility for the most part.

Staubli appears to claim that no changes should be needed. @gonzalocasas claims they are needed.

gonzalocasas, 2019-07-13:
This is what happens if you have a variable of type `screen` in your globals:
[image]

I will try a few ways to workaround this in the next days, but my impression is that it is not really possible unless the variables can be locals, and then the loading is conditional to the controller version (but again, not sure if that will work). The next best option imo would be to have two different `ros_server.dtx` files, and then the installation process involves choosing one or the other, maybe defaulting to the new one, and having the older CS8 compatible one around just in case.

gonzalocasas, 2019-07-14:
It looks like there's hope ;) I tried moving the code that is specific to a controller version into a libraries, and then load the correct one for the controller dynamically using `libLoad()` and that *seems* to work.

I'm still not sure whether I can really do all that's needed from a library (loading user pages and so on), but at least the problem of not being even able to load the app due to the unsupported `screen` type is gone.

gavanderhoorn, 2019-07-15:
Hm.

@gonzalocasas: would you see a possibility to give Staubli a call and ask them about the type issue directly?

I like dynamic loading of libraries as much as anyone, but if Staubli is so sure that things should-just-work (at least according to what @nilsmelchert tells us), perhaps we're missing something?

gonzalocasas, 2019-07-15:
Yep, of course, I'll do that tomorrow.

gonzalocasas, 2019-07-16:
I paste here verbatim the written answer of our Stäubli technical contact confirming the incompatibility:

```
„ all VAL3 code is compatible with both controller versions.” This statement is not correct. Most code is, but not all. Screen variable are not supported by CS9. As well as all the userpage instructions coming from CS8C and lower (title, cls, get, gotoxy, put, ...). On the other hand, the usepage instructions used on CS9 are not supported on CS8C either. One is a LCD screen based on char and the other is a multicolor touchscreen web based.
If you are using userpages, you are forced to write two applications.
```

nilsmelchert, 2019-07-16:
Either I got the wrong information or I couldn't get to the heart of the problem. In any case, I apologize for the wrong information.
As I said, I only have experience with the CS8 controller.

gonzalocasas, 2019-07-16:
@nilsmelchert no worries, it's anyway good to get official confirmation about this.

The next open question would be what approach we want to take:
  1. one app with dynamic lib loading
  1. two totally separate apps, one per controller
  1. one app + two folders with CS8 and CS9 specific code in them, and a deployment process that involves picking the right one and copying it to the controller
  1. something else?

gavanderhoorn, 2019-07-16:
@gonzalocasas wrote:
> @nilsmelchert no worries, it's anyway good to get official confirmation about this.

exactly, that was what I was after.

> The next open question would be what approach we want to take:
> 
>     1. one app with dynamic lib loading
>     2. two totally separate apps, one per controller
>     3. one app + two folders with CS8 and CS9 specific code in them, and a deployment process that involves picking the right one and copying it to the controller
>     4. something else?

we could refactor the code such that the display component is actually a stand-alone program, that gets its data from reading variables / exposed state from the actual driver.

I'm not sure that would be more efficient than just saying: "pick the correct version of *this* file" (where "this file" is the one that implements the display component).

It would have been nice if Staubli would have provided some bw-compatible way to reuse code which uses userpages (fi by mapping calls to their newer equivalents). But that was probably deemed not (economically) feasible or interesting.

gonzalocasas, 2019-07-16:
> we could refactor the code such that the display component is actually a stand-alone program

That's exactly what I started doing, and then load the program as a library with `libLoad()`

gavanderhoorn, 2019-07-17:
> > we could refactor the code such that the display component is actually a stand-alone program
> 
> That's exactly what I started doing, and then load the program as a library with `libLoad()`

Ah, ok.

Then I'd misunderstood you.

simonschmeisser, 2019-07-23:
Just a quick feedback, I installed the program on a TX2-90L and it runs nicely so far, thanks a lot!

simonschmeisser, 2019-07-25:
Another potential incompatibility (?)
https://github.com/gramaziokohler/staubli_experimental/blob/kinetic-devel-cs9/staubli_val3_driver/val3/ros_server/ros_server.dtx#L35

I had to change this from
```<Value key="0" fatherId="flange[0]" ioLink="valve1" />```
to
```<Value key="0" fatherId="flange[0]" ioLink="ValveIO/valve1" />```
in order to be able to toggle the valve by setting
```tTool.gripper = true ```

Not sure if that used to work on CS8 as I guess it was never tested there either

gavanderhoorn, 2019-08-01:
Interesting that @gonzalocasas didn't run into this.

gonzalocasas, 2019-08-01:
In our project we don't have any gripper, only static tools, so we never toggle these valve settings.

gavanderhoorn, 2019-08-01:
But you load the same files as @simonschmeisser, no?

simonschmeisser, 2019-08-01:
there is no "compile" error, just nothing happens. Maybe some error can be read, don't remember

marshallpowell97, 2019-11-25:
I can confirm that the only major change between Val3 rev. 7 (CS8C) and rev. 8 (CS9) is the handling of the UI. There was a minor change in the way that the IO was mapped (@simonschmeisser figured it out). This would normally be done automatically in SRS so wouldn't be considered a major change. Everything else should work the same. I will test on a CS9 just to confirm.

gavanderhoorn, 2019-11-26:
@marshallpowell97: thanks for your comment.

> There was a minor change in the way that the IO was mapped (@simonschmeisser figured it out). This would normally be done automatically in SRS 

but the controller doesn't do this, correct? If users would copy the `val3_driver` sources directly to the controller they would have to change it manually?

marshallpowell97, 2019-11-26:
That's correct.

simonschmeisser, 2019-11-26:
@marshallpowell97 do you know about a way to ``#ifdef`` those incompatibilities so that we can continue to have one code base for both CS8 and CS9 ?

marshallpowell97, 2019-11-26:
There isn't a super clean way to do it, but `getVersion("System")` will return a string with the version of SRC (the firmware on the controller). "s8. ..." will be CS9 and "s7. ..." will be CS8C. So something like:
```
  if(mid(getVersion("System"),1,1) == "7")
    lib:libLoad("CS8C")
  elseIf(mid(getVersion("System"),1,1) == "8")
    lib:libLoad("CS9")
  endIf
```
Would work. All of the UI and the IO would need to be moved to a library, but the main code base could be kept the same.

gonzalocasas, 2019-11-26:
Yup, that's what I tried back then: https://github.com/ros-industrial/staubli_experimental/issues/34#issuecomment-512037426

It worked fine, but we were not sure if that was the official Stäubli way to achieve it.

@gavanderhoorn should I go in that direction and send a PR then?

gavanderhoorn, 2019-11-26:
> should I go in that direction and send a PR then?

Yes, please.

With a PR we can discuss the approach.