ros-industrial/staubli_experimental #37: Too many joints/tool0 overconstrained

simonschmeisser, 2019-08-28:
i was investigating some strange self-collisions we got with our tx2-90l model and finally found that there are joints from both
```
link6 -> tool0
```
as well as
```
link6 -> flange ->tool0
```
eg for tx90 in https://github.com/ros-industrial/staubli_experimental/blob/1ff47df23612ba242b711e00ed3e5a3764eaec28/staubli_tx90_support/urdf/tx90_macro.xacro#L214 and (ff) https://github.com/ros-industrial/staubli_experimental/blob/1ff47df23612ba242b711e00ed3e5a3764eaec28/staubli_tx90_support/urdf/tx90_macro.xacro#L228

Same for at least tx-90L and tx-90XL (https://github.com/ros-industrial/staubli_experimental/commit/079b04a42b8c54d752e34082744af0126752faf9) and in main staubli repo rx160 and 160L (introduced in https://github.com/ros-industrial/staubli/commit/d51899f7a19f733f254b191a6f981778cc6c308b)

the first joint `link6-tool0` cannot be found in tx60, tx60L so everything is fine there

@gavanderhoorn @nilsmelchert I can do a PR but should I remove the first or the second one? I'm leaning towards keeping the direct joint as that means fewer transforms

simonschmeisser, 2019-08-28:
I suggest to fix it like this: https://github.com/isys-vision/staubli_experimental/commit/5cb02eb9fb3f0885759f5f065be2d34d0b23d096

gavanderhoorn, 2019-08-28:
Seems I missed that in the review of #22.

The correct sequence is: `link_6 -> flange -> tool0`.

See [Coordinate Frames for Serial Industrial Manipulators: flange](http://gavanderhoorn.github.io/rep/rep-0199.html#flange) and following sections for some rationale (wip, but I'd just be copy-pasting sections here).

nilsmelchert, 2019-08-30:
Yes, seems like I missed that.

closed 2019-08-30 by gavanderhoorn