ros-industrial/staubli_experimental #39: Strange accel override

simonschmeisser, 2019-09-02:
https://github.com/ros-industrial/staubli_experimental/blob/1355e50234063e5aed76f5e9a765c05080a72fef/staubli_val3_driver/val3/ros_server/pushMotion.pgx#L77

sets the acceleration (deceleration) to 9999 if velocity override is activated. This is motivated by a  reference to the manual and Val3's default value. However in said manual the acceleration is specified to be in percent, ie in the range of 100 to 0. 9999 is the default value for tcp velocity.

Our customer set this to 60 and obtained good results, 9999 was apparently very un-smooth and not visually pleasing

gavanderhoorn, 2019-09-02:
Unfortunately I wouldn't know why it's set to `9999`.

The driver was written by @muhrix and colleagues during his time at Ocado and he no longer works there. He currently works for a company that either doens't let him work on OSS or makes it (very) difficult to do so.

If `60` works, we could see whether that would make more sense as a default.

gavanderhoorn, 2019-11-23:
@simonschmeisser: have you had any further feedback from your customer?

simonschmeisser, 2019-11-23:
I heard no further complaints but I'll ask them and report back here

gavanderhoorn, 2019-11-23:
Thanks.

Performance of `staubli_val3_driver` ignoring the acceleration issue?

gonzalocasas, 2019-11-25:
@simonschmeisser thanks for the hint; I will give it a try on our setup. I also noticed the motions were rather un-smooth as you mention, but never put in the time to check why, and this could be it.

marshallpowell97, 2019-11-26:
I can confirm that accel is expressed as a percentage and has a default value of 100. There is essentially no upper limit on accel (up to 99999%) and putting in a value higher than 100% will cause the arm to attempt that acceleration.

gavanderhoorn, 2019-11-26:
Ok.

If we can come up with a better default, we could change the current value.