ros-industrial/staubli_experimental #43: Moving a real robot

lepalom, 2019-12-02:
I have tested this repo. I have build all the packages and installed the val3 driver with success.

I can communicate with the real robot and receive and check the position of the real robot. But, I'm blocked because I don't know ho to move the **real** robot. Please, could you write some lines, or some launch file showing how to move the real robot, for instance using joint_state_publisher?

gavanderhoorn, 2019-12-03:
> I can communicate with the real robot and receive and check the position of the real robot

May I then assume you've followed the steps in the readme of `staubli_val3_driver`?

And you can run the `robot_interface_streaming.launch` as indicated in the [Run the industrial_robot_client node (ROS-I client)](https://github.com/ros-industrial/staubli_experimental/blob/b4a0d23e54c7f48b5f043314517d782e1cc344de/staubli_val3_driver/README.md#run-the-industrial_robot_client-node-ros-i-client) section?

> Please, could you write some lines, or some launch file showing how to move the real robot, [..]

Almost all (serial/industrial) robot drivers in ROS expose a [FollowJointTrajectory](http://docs.ros.org/api/control_msgs/html/action/FollowJointTrajectory.html) action server. 

> for instance using joint_state_publisher?

`sensor_msgs/JointState` is only used to report current state of the robot. Never to command it. And `joint_state_publisher` is not used either.

---

To move a robot (real or simulated), you'd have to start up a `FollowJointTrajectory` client.

One well known one would be MoveIt, when configured with a `MoveItSimpleControllerManager` as the 'controller' (in quotes as it really just forward the computed plan to the external action server).

Alternatively you could write a short Python script and have that start a client and submit a `JointTrajectory`. We don't have an example here in this repository, but here is one for `motoman_driver`:

<details>
<summary>Example FollowJointTrajectory client Python script</summary>

```python
#!/usr/bin/env python

# TODO: add license

# A simple example showing the use of an Action client to request execution of
# a complete JointTrajectory.
#
# Note: joint names must match those used in the urdf, or the driver will
#       refuse to accept the goal.

import rospy
import actionlib
import time
import math

from control_msgs.msg import FollowJointTrajectoryAction, FollowJointTrajectoryGoal
from sensor_msgs.msg import JointState
from trajectory_msgs.msg import JointTrajectory, JointTrajectoryPoint

def main():
    rospy.init_node('simple_trajectory_action_client')

    # create client and make sure it's available
    client = actionlib.SimpleActionClient('joint_trajectory_action', FollowJointTrajectoryAction)
    rospy.loginfo('Waiting for driver\'s action server to become available ..')
    client.wait_for_server()
    rospy.loginfo('Connected to trajectory action server')

    # setup simple goal
    goal = FollowJointTrajectoryGoal()
    goal.trajectory = JointTrajectory()

    # this should correspond to the joint names of the robot being used (ie:
    # the joint names as specific in the urdf). The list specified here contains
    # the default Motoman joints.
    # NOTE: order matters here
    goal.trajectory.joint_names = ['joint_s', 'joint_l', 'joint_u', 'joint_r', 'joint_b', 'joint_t']

    # motoman_driver only accepts goals with trajectories that start at the
    # current robot state, so retrieve that and use it as the first point in
    # the trajectory
    robot_joint_states = rospy.wait_for_message('joint_states', JointState)

    # make sure the state we get contains the same joints (both amount and names)
    if set(goal.trajectory.joint_names) != set(robot_joint_states.name):
        rospy.logfatal("Mismatch between joints specified and seen in current "
            "JointState. Expected: '{}', got: '{}'. Cannot continue.".format(
                ', '.join(robot_joint_states.name),
                ', '.join(goal.trajectory.joint_names)))
        sys.exit(1)

    # create the trajectory: it will consist of three points:
    #
    #   1. T= 0: current pose
    #   2. T= 5: T joint rotated -20 degrees
    #   3. T=10: T joint rotated +20 degrees

    q0 = robot_joint_states.positions
    q1 = list(q0)
    q2 = list(q0)

    # Assume the last joint is T
    q1[-1] -= math.radians(20)

    # Make the robot come to a complete stop at each trajectory point (ie:
    # zero target velocity).
    qdot = [0.0] * len(goal.trajectory.joint_names)

    # add points
    goal.trajectory.points.append(JointTrajectoryPoint(positions=q0,
        velocities=qdot, time_from_start=rospy.Duration( 0.0)))

    goal.trajectory.points.append(JointTrajectoryPoint(positions=q1,
        velocities=qdot, time_from_start=rospy.Duration( 5.0)))

    goal.trajectory.points.append(JointTrajectoryPoint(positions=q2,
        velocities=qdot, time_from_start=rospy.Duration(10.0)))

    # goal constructed, submit it for execution
    rospy.loginfo("Submitting goal ..")
    client.send_goal(goal)
    rospy.loginfo("Waiting for completion ..")
    client.wait_for_result()
    rospy.loginfo('Done.')

if __name__ == '__main__':
    main()
```

</details>

This shouldn't work for a Stäubli robot, but it should get the idea across.

> I don't know ho to move the real robot

please realise btw that the `FollowJointTrajectory` action interface is generic. It's not restricted to real robots, or to any specific brand. Many drivers expose this interface, and Gazebo (with `gazebo_ros_control` fi) does as well.

gavanderhoorn, 2019-12-03:
As this is not an issue with the packages in this repository, I'm going to close it.

Feel free to keep commenting on it of course.

closed 2019-12-03 by gavanderhoorn

nilsmelchert, 2019-12-04:
@lepalom it could also be a hardware "problem". If you want to work with a real robot, you have to set it to the external mode first. You do this by turning the key to the correct position on the emergency stop provided. Furthermore you have to make sure that the brakes of the robot are released.

lepalom, 2019-12-04:
> May I then assume you've followed the steps in the readme of `staubli_val3_driver`?

Yes, and they worked.

> And you can run the `robot_interface_streaming.launch` as indicated in the [Run the industrial_robot_client node (ROS-I [...]

Yes, and when I move the robot manually the rviz is updated.

> Almost all (serial/industrial) robot drivers in ROS expose a [FollowJointTrajectory](http://docs.ros.org/api/control_msgs/html/action/FollowJointTrajectory.html) action server.

So, may I understand that I have to send a complete trajectory to the robot? or, could I send positions/velocities every 4 ms as val3 programs can do?

The UR driver let you send positions, and I'm asking if the val3 driver has similar functionality.

In any case, thanks for the reply.

lepalom, 2019-12-04:
@nilsmelchert thanks, just a misunderstanding with ros industrial features.

gavanderhoorn, 2019-12-04:
> So, may I understand that I have to send a complete trajectory to the robot? or, could I send positions/velocities every 4 ms as val3 programs can do?

`staubli_val3_driver` does not support position streaming at this point.

It's trajectory based.

From a look through the sources there don't appear to be any blockers to add that functionality, it's just that no one has done it yet.

> The UR driver let you send positions,

Both `ur_modern_driver` and `ur_robot_driver` are `ros_control` based and internally use a rather different approach to interfacing with the robot.

The two approaches cannot really be compared.

lepalom, 2019-12-04:
> From a look through the sources there don't appear to be any blockers to add that functionality, it's just that no one has done it yet.

So, should I make an issue to add this functionality? ;-) and yes, probably you will answer or make a PR adding it.

Sending point by point a trajectory is an interesting feature.

gavanderhoorn, 2019-12-04:
Yes, please do.

I'm not entirely sure, but it could be that both `staubli_val3_driver` and `industrial_robot_client` need to be changed to support this.