ros-industrial/staubli_experimental #45: Velocity Override does not have correct effect

marshallpowell97, 2020-01-10:
https://github.com/ros-industrial/staubli_experimental/blob/b4a0d23e54c7f48b5f043314517d782e1cc344de/staubli_val3_driver/val3/ros_server/pushMotion.pgx#L72-L79
Sets the velocity to the value set on the teach pendant if 'velocity override' is enabled. This does not seem to work as intended since the velocity set on the pendant already "overrides" whatever value is set in the Val3 code. The result is that the "overridden" velocity ends up being much smaller than intended.

I.E. if 50% velocity is set by the pendant, the actual speed of the robot will be 50% (result from `getMonitorSpeed()` ) multiplied by another 50% (set by the pendant) = 25%

gavanderhoorn, 2020-01-10:
I'm not entirely sure what this code was supposed to achieve either.

This was all added in #10 and unfortunately the contributor of that is no longer around.

I suggest we remove this, and the associated code in the UI files.

@nilsmelchert @simonschmeisser: have you ever used this and did it work for you as expected?

simonschmeisser, 2020-01-11:
Unfortunately we currently don't have access to a Stäubli and I went on summer holidays after getting the thing to run so I haven't collected that much experience but ...

I image the original intention, which might have been achieved or not, was to actually speed up execution of trajectories? At least that would be problem we keep pushing ahead and not addressing. Basically if you set velocity levels and acceleration levels too low and/or your time parametrization is suboptimal you end up with a trajectory containing values smaller than the maximum achievable velocity. Then the controller will interpret those as upper limits, reducing velocity further. Now thinking about it it is however not clear to me why we would not "solve"* this at the MoveIt level ... 

* or somewhere else external to the driver