ros-industrial/staubli_experimental #50: TX40 gazebo

hehonglu123, 2020-05-13:
Hi all,

Is the model for Staubli TX40 available in gazebo?
@johnwason

gavanderhoorn, 2020-05-13:
All the packages we have here are what you can see in the repository (and in `ros-industrial/staubli`).

So no, right now we don't have a support package nor Gazebo support for the TX40 here.

gavanderhoorn, 2020-05-13:
I just remembered I had seen at least a support package somewhere recently, and found it: [Tomiliat/Staubli_backup/staubli_ws/src/staubli_tx40_support](https://github.com/Tomiliat/Staubli_backup/tree/0932e66c12cec5fc2774c96f63d895cdcdeb9b07/staubli_ws/src/staubli_tx40_support).

Note: I haven't used this myself, nor have I verified it is correct.

I cannot say anything about the state of it.

hehonglu123, 2020-05-13:
@gavanderhoorn 
Thanks for pointing it out, I'll try it out.

closed 2020-05-13 by hehonglu123

gavanderhoorn, 2020-05-14:
@hehonglu123: what was the conclusion here?

hehonglu123, 2020-05-14:
The urdf seems to have collision problem on my side in gazebo, it loads fine after I commenting out the collision element. It can be loaded to tesseract with collision element directly.

hehonglu123, 2020-05-15:
I just found out it doesn't quite look like the actual tx40, and the model is always spawned in the world center and not able to drag it around. 
[image]
[image]

gavanderhoorn, 2020-05-16:
That doens't look good, no.

Seems you'll have to create the support package yourself then.

thanks for letting us know.

gavanderhoorn, 2020-05-16:
Oh and if you do end up creating a support package for the TX40, please consider submitting a PR to add it here to the repository.

sinamr, 2020-09-10:
There is a normal version which be found here:
https://github.com/Tomiliat/Staubli_backup/tree/master/staubli_ws/src/staubli_tx40_support/meshes/visual/normal
I don't know how arcuate the model is , but .stl files seem visually fine.