ros-industrial/staubli_experimental #55: Ways to contribute to this Project

Wifi-cable, 2021-09-23:
Hello,
As requested on ROS-Discourse I write an issue on the Github Repro of this driver. Currently I'm looking to learn more about every aspect of ROS, since working with ROS and this driver in particular  will soon be part of my job. I would like to be able to do something, that benefits the Robot community as a final year project. (Bachelors Thesis)  I was suggested a wide range of topics. (not all are related to this driver) 

I'm currently looking for feedback regarding those ideas: 
- Migrate this driver to ROS2?
- Build a "fool prove GUI" prototype for this Robot, to let non tech people "teach" the robot new skills?

Are there other Ideas how I could help, given I have 3 month time and limited experience?

gavanderhoorn, 2021-09-28:
Migration to ROS 2 is an option, but please note: this is not a `ros_control` based driver. It's using `industrial_core`, which itself has no ROS 2 equivalent yet.

We have been discussing with various stakeholders how we could best continue development here:

 - create a new driver, potentially using UniVAL
 - continue the VAL 3 based approach we have here

the first option might be good for certain applications, where the external entity (ie: a ROS node / application) is in complete control. The downside is that it would require a UniVAL license, which is not free. It also removes the possibility to easily create a setup where autonomy or control is shared (switching to UniVAL is a boot-time affair, you can't do it at runtime).

Continuing development of the VAL 3 based driver is largely independent of whether ROS 1 or ROS 2 is used. Most of the improvements would be on the VAL 3 side (ie: the code which runs on the controller).

There are certainly things to improve there, as is evidenced by the list of open issues we have. More direct control over the motion, with better velocity control (note: *not* velocity control, but better control over the velocity while executing a trajectory) would be a good first step.

If you're really looking to do something with ROS 2 and `ros2_control`, you *could* see whether implementing a hardware interface which 'speaks' Simple Message performs sufficiently. It would not solve the problems with trajectory execution (as that would be on the VAL 3 side), but it would help with ROS 2 compatibility.

> Build a "fool prove GUI" prototype for this Robot, to let non tech people "teach" the robot new skills?

this sounds like a robot-agnostic kind of thing. It would not be directly tied to Staubli robots, nor benefit the packages specific to Staubli directly.

gavanderhoorn, 2021-09-28:
Note, when I refer to "the VAL 3 driver", I'm talking about [ros-industrial/staubli_val3_driver](https://github.com/ros-industrial/staubli_val3_driver).